alienatedsec/solis-ha-modbus-cloud #2: Wiring diagram

matthiasdeblock, 2023-03-16:
Hi

thank you for this GH page, it looks amazing!!! Nice work!!! I am currently in the same situation as you are with my Solis inverter and the combination Home Assistant / SolisCloud. 

I was looking at your setup but the only thing that is not really clear for me is the wiring... How did you do the wiring? Is it possible to add a photo or something like that of the wiring? 

Thank you!

Regards
Matthias

alienatedsec, 2023-03-16:
Hi @matthiasdeblock 

Yes, I was thinking about it. Sometimes when you think about the bigger picture, details are missed. I will make it available once created.

solarjones, 2023-03-17:
I've just purchased 2x Waveshares, connector, POE switch and DIN rail as per your project. Like many others, want the Solis Cloud and HA connectivity.  Thanks so much for sharing this information and look forward to the pin out wiring details.  Also, photos of your finished project would be great.

Thanks in advance!
Simon

alienatedsec, 2023-03-17:
Done. Enjoy and share with the community @solarjones @matthiasdeblock

closed 2023-03-17 by alienatedsec

wymes, 2023-07-03:
Hi

Firstly great article and it clarified a few things for me re HA connectivity . Would it be possible to achieve the same thing if I used a single (USR-W610 Serial to WiFi Ethernet Wireless Converter RS232 RS485 Serial Server Support WatchDog Modbus Gateway TCP UDP) to replace the waveshare devices 

In essence I want to still use the s3 dongle if possible and use the 610 to integrate the inverter with with HA . I am not quite clear how the 485 wiring would be between the inverter  - W610 - S3 dongle 

Regards 

Nigel