alienatedsec/solis-ha-modbus-cloud #13: Modbus stops working to HA when battery turned on.

MatcherMike, 2023-05-21:
Hi,

Firstly, many thanks gor helping with my ha integration on the other thread. I'm happy I get figures except the ct pv generation value.

I have replicated your setup. Thanks for the great detail. I get values into HA, except, when I turn the power switch on my battery on! I then get no new values. For example, Meter active power stays old and the minutes since update count up. There is little to no flashing of the activity blue led.

Turn off the battery and within 30seconds the values update. 

Any ideas please?

Inverter is solis ac.

Thanks
Mike

alienatedsec, 2023-05-21:
@MatcherMike 
I am not sure which option you activate and which is not working - please can you paste any screenshots? Also, I would need more about your setup i.e. Inverter model and which integration is used.

MatcherMike, 2023-05-31:
Hi, sorry for the delay in replying. I still don't have this working and have been running with the datalogger direct into the inverter. I have today conducted some further testing.

Inverter: SN 060262225140105
Type: rai-3k-48es-5g
Operation: Grid connected AC battery only mode (Not Hybrid)
Battery: Seplos BMS with 16x Cells
Datalogger:  S3-WIFI-ST
HA Version: 2023.5.4, Supervisor 2023.04.1, Operating System 10.2
HA Integration: https://github.com/wills106/homeassistant-solax-modbus, Using Solis inverter type - Version 2023.05.2

If i have my battery system (seplos BMS) turned off, then i get soliscloud stats ok. I also have stats into HA using the above integrations:

[image]
[image]
[image]

If i turn my battery system on, it all stops working:
[image]
[image]

Stats in solis cloud remain static, until they eventually disappear.

Here is the connectivity of my system:

[image]
[image]
[image]
[image]
[image]

I have no idea why turning the battery on, would stop it all working!
My solis modbus integration does still complain about the serial number of the inverter, but it pulls stats into HA ok.

Happy to take more screenshots, and to try anything. Really appreciate your help with this.
Mike

alienatedsec, 2023-05-31:
@MatcherMike is the BMS IP connected? If so, it may have the same IP as your TCP Server and conflict when turned on at the same time.

MatcherMike, 2023-05-31:
> @MatcherMike is the BMS IP connected? If so, it may have the same IP as your TCP Server and conflict when turned on at the same time.

Good question. No it has no IP connectivity. Just the RS485 CAN connection to the Inverter. I currently manage it via BT.

I have previously connected to it via RS485 adapter for adjusting some settings.

alienatedsec, 2023-05-31:
What settings? Are those configurable on both, the inverter and the BMS?

MatcherMike, 2023-05-31:
I had to set the communications type on the BMS and the inverter to get the CAN working. These are the only changes

Seplos documentation is here: https://drive.google.com/drive/folders/10pxgNLHovcDZRVGrCZsSkfecBrRw-AdW

MatcherMike, 2023-05-31:
I think i may have resolved this now - I removed my earth grounding for the two waveshare units, and i now have stats through HA modbus and solis cloud. Not sure why this would cause an issue.

Will test further through tomorrow,
Thanks

alienatedsec, 2023-05-31:
> earth grounding

Which one? Earth and ground has different connections and technically both are slightly different. 

https://byjus.com/physics/difference-between-earthing-and-grounding/

Where did you connect your earth (green) cables?

MatcherMike, 2023-05-31:
Pgnd, the leftmost connector on the top of the waveshare. I had this connected to an electrical earth point. Same earth point as my inverter and battery system.  Confused as it worked normally, but not with battery on.

Where did you connect yours?

Thanks

alienatedsec, 2023-05-31:
> Where did you connect yours?

There is an earth rod behind my garage. The same connected to my inverter's radiator. I connected it directly there.

MatcherMike, 2023-05-31:
That's effectively what i had her. No worries, seems like it isn't necessary and hopefully continues to work.

Many thanks for responding, and many thanks for putting together this great guide.

closed 2023-05-31 by MatcherMike