alienatedsec/solis-ha-modbus-cloud #18: TCP Client & Server - S2-WL-ST Troubleshooting and Config changes

alienatedsec, 2023-06-12:
### Discussed in https://github.com/alienatedsec/solis-ha-modbus-cloud/discussions/14

<div type='discussions-op-text'>

<sup>Originally posted by **solarjones** May 21, 2023</sup>
I've been running 2x Waveshares together with a per the design for just over 2 months and during that time have had a few issues.

The Solis data stick I've used since installation (S2-WL-ST) with Solis Cloud had been flawless during that time until very recently where it went offline. I power cycled my PoE hub which brought it back online briefly, but in the end had to unscrew the data stick to power it off entirely (which makes sense since it would still receive power directly from the inverter).  This is the first time the Solis data stick has 'hung' since installation.

Previously the anomalies have been with the HA data spiking on battery SOC, export & import values, which in turn produces bad energy stats.

I'm running Solax Modbus 2023.05.2b3 with the following configs on Waveshare:

[image]

[image]

Server
[image]
[image]

Client
[image]
[image]

The pattern I've observed is everything works well for a few weeks then hits an issue. 
Any suggestions?    
Thanks.
</div>

alienatedsec, 2023-06-12:
Hi @solarjones

While this has been discussed for over a month, I wasn't sure where is the issue and if this is something we should be mitigating. My initial testing with new firmware `v1.486` and the Modbus gateway type amended to `multi-host non-storage` type makes a massive difference to performance.

Can you please confirm this is also working for you?

solarjones, 2023-06-12:
I've not loaded the new firmware as yet as I was waiting to see if the other previous parameter tweaks improved things. Dare I say it, but I think it has, as so far I've had no SOC drops or spikes (now I said it, I await it!).

alienatedsec, 2023-06-12:
@solarjones have you just changed the gateway type? or what other parameter tweaks and improvements do you have in mind?

solarjones, 2023-06-12:
No I haven't changed the gateway type as yet, only the client configs according to your [previous](https://github.com/alienatedsec/solis-ha-modbus-cloud/discussions/14) instructions.

I'll try the firmware updates one evening this week. Assuming it resets all configs, is there a known good set of config values to use? 
Thanks

alienatedsec, 2023-06-12:
@solarjones if you could try the below setting on `TCP Server` using the old firmware, it would be great.

[image]

alienatedsec, 2023-06-17:
@solarjones I have not experienced any data spiking since 9th June after following the latest [configuration guideline](https://github.com/alienatedsec/solis-ha-modbus-cloud#configuration) 

If yours are still ongoing, please follow the same.

closed 2023-06-17 by alienatedsec

solarjones, 2023-06-17:
Touch wood, all good for me. However I've not got round to updating the Waveshare firmware yet so my configuration doesn't match your guide currently. I'll keep a watch on things and check configs etc. Thanks for your diagnostic work!

solarjones, 2023-06-26:
This evening it spiked again. I should point out I've yet to apply the latest firmware as per your guide, but to date, since the last config tweaks we'd previously discussed, all has been well. Until....

I noticed the Backup lightbulb icon was not on the inverter display. Very odd because I'd definitely enabled Backup mode. So I went into the inverter setup, under Storage Control Parameter, and enabled 'Backup'. Within 10 seconds of doing this, I got corrupt data (inspired by one of your previous comments, I've setup an automation to message me when SOC hits zero which has been a constant in this saga of spiky data).  To recover, I simply reloaded the Solax plugin. I did not reboot the Waveshares. 

This is either a coincidence (doubtful) or enabling Backup mode on the inverter has caused an issue. I also noticed that Backup mode is getting disabled somehow (not by me). I didn't touch any cabling at the time of the Solis parameter change so very puzzled by this.

alienatedsec, 2023-06-26:
@solarjones mine has been stable since the changes I made. Despite a newer firmware (likely related) the key thing for me is the `Multi-host non-storage type` on the `TCP Server`

[image]

I am also on a higher baud rate just for the `TCP Server` and the inverter matching `38400`. My `S2-WL-ST`, so the `TCP Client` is still on `9600`.

[image]

@jevburchell - Solis confirmed (or maybe just tried to close my ticket) that there is no way to change the baud rate on newer dataloggers `S2-WL-ST` and `S3-WIFI-ST`.