alienatedsec/solis-ha-modbus-cloud #44: Can RS485 port be used in Solis Inverter ?

jan-dane, 2025-03-29:
Can we use the built-in RS-485 port for data extraction for monitoring purpose as shown in pic below (Solis Inverter) instead of  using Data logger ?

[image]

[image]

alienatedsec, 2025-03-29:
Yes. You can even find the pinouts and use only one Waveshare for this.

jan-dane, 2025-03-29:
[image]

Taken from Solis Manual.

eqasxar, 2025-04-16:
It's very disappointing, but I can't connect to the new version of the Solis S6 Hybrid Inverter. I contacted support. My inverter has had its firmware updated remotely. The port doesn't respond to any requests. I'm providing an RS485 from my inverter, a support-approved RJ45 socket.
Maybe someone has an idea how to connect to the same USB port that the wifi dongle uses. Back to the question https://github.com/alienatedsec/solis-ha-modbus-cloud/issues/43

[image]

ebabenskas, 2025-04-17:
@eqasxar  What did Solis support suggest? Do they have any explanation for why it’s not working?

I also have a Solis S6 with the S2-WL-ST USB. I previously tried using a simple RS485-to-USB adapter and attempted to connect directly to Home Assistant using the homeassistant-solax-modbus add-on. It wasn’t successful, but I thought the issue was due to my cheap USB adapter.

I ended up buying a Waveshare adapter, but haven’t had time to test it yet. If it doesn’t work either, that would be really disappointing. :/

Regarding the S2-WL-ST USB — I think it acts only as a connector. If you look inside the inverter, you can see the RS485 A/B cables going from the USB socket. So, if the RJ45 port doesn’t work, I plan to buy another Waveshare adapter along with a USB male/female connector, and I believe it should be possible to wire it up that way too

alienatedsec, 2025-05-30:
> Do you think this is possible?

@uppaljs Possible as there are cables to convert USB to RS485 already available https://www.youtube.com/watch?v=cq5_S6no7cw

I have seen another over here https://shop.libra.energy/en_GB/product/solis-usb24p-adapter

uppaljs, 2025-05-30:
Hi, Thanks for the quick reply , I saw that, but was unable to find it anywhere to buy on the internet. If I can get a pinout / diagram, I can probably build a connector / usb cable but I have no way of knowing without a diagram or buying this USB to 4pin converter.

josu98, 2025-07-23:
> [@eqasxar](https://github.com/eqasxar) What did Solis support suggest? Do they have any explanation for why it’s not working?
>
> I also have a Solis S6 with the S2-WL-ST USB. I previously tried using [...]

I’m using an S6 hybrid inverter with an S2-WL-ST USB adapter connected to my local LAN. I have a Waveshare device configured as a Modbus server, directly connected to the RS485 port on the inverter. The Waveshare unit is also connected to the LAN.

My Home Assistant server uses the Solis Modbus integration to read data from and control the inverter locally. With this setup, I’m able to use both Solis Cloud (https://www.soliscloud.com) and Home Assistant simultaneously without any issues. They've been running in parallel smoothly.